gkaindl/ambi-tv #83: Raspberry PI 2

ghgoldberg, 2015-02-12:
Has anyone implemented ambi-tv on a RPi 2?  I am going to order an RPi2 to REplace my RPi 1, model B. Any ideas as to whether it would boot from the same SD card without any changes?   I would imagine that there would (at the very least) be some driver issues?

G-

zeugri, 2015-02-17:
Hi,

i'm trying to do this, from a B+ to a RPi2. For the moment i can boot on the sd card, i can compile the ambi-tv sources and run the binary.

but i get a compilation error for the usbtv-driver thus far.

```
pi@rpi:~/ambi-tv/ambi-tv/misc/usbtv-driver$ make
make -C /lib/modules/3.18.7-v7+/build  M=/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver modules
make[1]: Entering directory '/usr/src/linux-source-3.18.7-v7+'
  CC [M]  /home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.o
/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.c:638:8: warning: initialization from incompatible pointer type [enabled by default]
/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.c:638:8: warning: (near initialization for ‘usbtv_vb2_ops.stop_streaming’) [enabled by default]
/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.c: In function ‘usbtv_probe’:
/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.c:719:8: error: ‘V4L2_FL_USE_FH_PRIO’ undeclared (first use in this function)
/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.c:719:8: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:263: recipe for target '/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.o' failed
make[2]: *** [/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver/usbtv.o] Error 1
Makefile:1381: recipe for target '_module_/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver' failed
make[1]: *** [_module_/home/pi/ambi-tv/ambi-tv/misc/usbtv-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-source-3.18.7-v7+'
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
```

i'm trying on a brand new install to see if it's better

zeugri, 2015-02-19:
After a bit more search on the usbtv topic i came across this info on the LinuxTV website (http://linuxtv.org/wiki/index.php/Easycap) : PAL support is built in the Linux kernel starting with 3.14, at least for my device (Fushicai video grabber ,ID 1b71:3002).

So, at this moment my Kernel is 3.18.7-v7+ on RPi 2 (raspbian). ambi-tv works directly without needing the custom usbtv driver.

Thanks for this great piece of software.

ghgoldberg, 2015-03-19:
@Zeugri -
Thanks for the posts.  

I am having a difficult time in distinguishing the commands and edits associated with the USB driver modifications (performed using the Fushicai with the Raspian Wheezy 2013-07-26 on the RPi 1) from those commands and edits needed to get ambilight to run on the RPi 2,

@Angora posted the following in issue #48:

<----------- clip
sudo apt-get install usbutils
wget https://github.com/gkaindl/ambi-tv/archive/master.zip
unzip master.zip
cd ambi-tv-master
cd misc
sudo bash get-kernel-source.sh
cd usbtv-driver/
make
sudo mkdir -p /lib/modules/3.6.11+/extra
sudo cp usbtv.ko /lib/modules/3.6.11+/extra
sudo depmod 3.6.11+
sudo modprobe videobuf2_core
sudo modprobe videobuf2_vmalloc
sudo modprobe usbtv

change the file:
sudo nano /etc/modprobe.d/raspi-blacklist.conf
to:
#spi-bcm2708

insert into:
sudo nano /etc/modules
these lines of code:
spi-bcm2708
videobuf2_core
videobuf2_vmalloc
usbtv

<------ clip

What lines are the ones needed for installing ambilight running with the current build of Raspdian on the RP1 2?

G-

patrickmollohan, 2017-09-04:
Sorry for digging up an old issue, but I did just get this running perfectly on a Pi 2. I used the 2015-01-31 build of Raspbian (found at: https://downloads.raspberrypi.org/raspbian/images/raspbian-2015-02-02/) and updated the kernel to 4.05 via the command  `sudo rpi-update 6d25078b7cb361052ae280d552561d7250d33f5b`
This kernel not only supports video grabbing from the Fushicai devices, but audio as well, which comes in handy if you want to check out xSnowHeadx's fork, which includes a spectrum analysis mode. Finally, make sure to enable the SPI bus via `sudo raspi-config` under "Advanced Options".