gkaindl/ambi-tv #85: Any idea what's happening here?

heteroerectus, 2015-02-18:
The whole system quit working, so I redid all the wiring, but now I'm getting a weird blinking. With two 3A power supplies wired in parallel, I can get it to rainbow mode, but with the single 5A supply I can't. The HDMI behavior is the same. 

Anyone see anything like this?
https://www.youtube.com/watch?v=lo87vAq0iTk

Mrjohns42, 2015-02-18:
It's categorically a bad idea to have voltage sources in parallel.

heteroerectus, 2015-02-18:
I've heard that, but the guys in my lab at Intel do it all the time for quick tests. It was showing the exact same behavior with 3A as well after the Pi startup sequence, so I don't think anything is fried.

ghgoldberg, 2015-02-18:
I second what MrJohns said.  Use a single power supply AND COMMON GROUNDS!  This has been mentioned numerous times in this forum.  If you do not use common grounds, you are wasting your time (and ours for asking).

 

G-

heteroerectus, 2015-02-18:
It is hooked up to a single power supply but the behavior hasn't changed. I am using common grounds on everything. 

Is there anything else I can try?

Mrjohns42, 2015-02-18:
How many volts is your power supply?

ghgoldberg, 2015-02-18:
I used a Mean well RS-75-5 5V 12 amp power supply

 

You can read about my hardware implementation in Issue #48.  Includes pictures, one of which is reproduced below:

 

 

G-