gkaindl/ambi-tv #86: Almost there but getting crazy disco light show

beefsnaps, 2015-04-22:
Hey,

I'm painfully close to getting this sorted and would appreciate some help. This is what I've done (I've tinkered a bit so hopefully I can remember everything):

- Got an RPI B, LPD8806 LEDs, Fushicai USBTV007 USB grabber connected to component out of Sky HD box
- Power from a 5v 4a adapter that powers LEDs and RPI - **do I need more power?**
- LEDs are 100% in the right PINs on RPI with the extra GND connected to the pin next to the PIN the SPIs are connected to. Power going to the plug with the RPI - **is this the correct for the GND? I've heard this could be an issue**
- Followed all the software steps here: https://github.com/gkaindl/ambi-tv/issues/7#issuecomment-23190594 - doing a lot of this via SSH on my Macbook as it is easier than hunched next to my TV trying to read the tiny text
- Didn't get the Video0 initially so bought a powered USB hub and bingo
- ran mplayer through ssh and got nothing on my mac but via TV I had a weird flashing image of the TV feed and command prompt alternating/flashing, there was lots of glitches too which might not be helping - I overclocked to highest setting but still got horizontal lines. When I run mplayer from RPI directly I just get a blank screen
- Turned on SPI in ambi-conf (no one tells you that!) Got spidev0.0 in /dev/ but when I ran the Ambi-TV bin I got a permission error: *lpd8806-spidev: failed to open device '/dev/spidev0.0' : 13 (Permission denied)*
- Dug around and found this command that seemed to solve the problem: *sudo chmod a+rw /dev/spidev0.0* - **no guides mention this step so could be a bigger problem somewhere? Also I need to do this every boot which is worrying**
- I havent cut and soldered the LEDs just kept them on the 5m roll to test. The first 18 or so lights are on permanently with different colours - **could potentially be an issue but didnt want to chop up my reel and it didnt work**
- When I run the bin and it works but I get what looks like the TV (the colours are a bit off but I'm sure just that is the settings) but it is flickering a bit and those first 18 lights go crazy

Any help would be appreciated! Let me know if you need any more info.

Cheers!

ghgoldberg, 2015-04-22:
I like that you are thinking about the grounds.  It is important that the RPi, the LEDs, and the power supply all are grounded AT A SINGLE POINT. I used a barrier strip at my power supply. 4 Amps would be ok for a short strip. My strip is something like 105 LEDs long (for a 65" monitor), and I use a 12A 5VDC supply. Also for even power to the LEDs, you should connect the power to BOTH ends of the strip. 

That's all I can offer now, i'm not at home, so I can't look up details.

Sent on the new Sprint Network from my Samsung Galaxy S®4.

<div>-------- Original message --------</div><div>From: beefsnaps <notifications@github.com> </div><div>Date:04/22/2015  5:58 PM  (GMT-05:00) </div><div>To: gkaindl/ambi-tv <ambi-tv@noreply.github.com> </div><div>Subject: [ambi-tv] Almost there but getting crazy disco light show (#86) </div><div>
</div>Hey,

I'm painfully close to getting this sorted and would appreciate some help. This is what I've done (I've tinkered a bit so hopefully I can remember everything):

Got an RPI B, LPD8806 LEDs, Fushicai USBTV007 USB grabber connected to component out of Sky HD box
Power from a 5v 4a adapter that powers LEDs and RPI - do I need more power?
LEDs are 100% in the right PINs on RPI with the extra GND connected to the pin next to the PIN the SPIs are connected to. Power going to the plug with the RPI - is this the correct for the GND? I've heard this could be an issue
Followed all the software steps here: #7 (comment) - doing a lot of this via SSH on my Macbook as it is easier than hunched next to my TV trying to read the tiny text
Didn't get the Video0 initially so bought a powered USB hub and bingo
ran mplayer through ssh and got nothing on my mac but via TV I had a weird flashing image of the TV feed and command prompt alternating/flashing, there was lots of glitches too which might not be helping - I overclocked to highest setting but still got horizontal lines. When I run mplayer from RPI directly I just get a blank screen
Turned on SPI in ambi-conf (no one tells you that!) Got spidev0.0 in /dev/ but when I ran the Ambi-TV bin I got a permission error: lpd8806-spidev: failed to open device '/dev/spidev0.0' : 13 (Permission denied)
Dug around and found this command that seemed to solve the problem: sudo chmod a+rw /dev/spidev0.0 - no guides mention this step so could be a bigger problem somewhere? Also I need to do this every boot which is worrying
I havent cut and soldered the LEDs just kept them on the 5m roll to test. The first 18 or so lights are on permanently with different colours - could potentially be an issue but didnt want to chop up my reel and it didnt work
When I run the bin and it works but I get what looks like the TV (the colours are a bit off but I'm sure just that is the settings) but it is flickering a bit and those first 18 lights go crazy
Any help would be appreciated! Let me know if you need any more info.

Cheers!

Mrjohns42, 2015-04-22:
Those lights draw about 1A per meter.  If you haven't cut the light strip to size and are connecting the whole roll, it's likely you're not supplying enough current for the LEDs to function properly.

ghgoldberg, 2015-04-22:
I agree.  I used the 12A supply for my 5m strip. Also powers the RPi, splitter, HDMI to AV conv, and USB converter. Better to have more capacity,  than not enough.

Sent on the new Sprint Network from my Samsung Galaxy S®4.

<div>-------- Original message --------</div><div>From: Mrjohns42 <notifications@github.com> </div><div>Date:04/22/2015  6:33 PM  (GMT-05:00) </div><div>To: gkaindl/ambi-tv <ambi-tv@noreply.github.com> </div><div>Cc: "Geoffrey H. Goldberg" <GHG@GeoffreyGoldberg.com> </div><div>Subject: Re: [ambi-tv] Almost there but getting crazy disco light show (#86) </div><div>
</div>Those lights draw about 1A per meter. If you haven't cut the light strip to size and are connecting the whole roll, it's likely you're not supplying enough current for the LEDs to function properly.

beefsnaps, 2015-04-24:
Thanks for the comments everyone.

 I've cut and attached my LEDs (power connected in a loop) and there is no flickering. I'm still using the old power supply and waiting for a 5v 12a one from Amazon to be delivered.

Still getting flashing random lights from the first 18 leds. Will feedback after I've added the new supply to see if that is fixed.

One thing software based, once I've run ambi-tv and quit, the program won't reopen as it says it is out of memory and I have to reboot. Not sure if anyone else has experienced that? I can't help thinking my issue is software/hardware as apposed to power supply.

Cheers!

Mrjohns42, 2015-04-24:
What do you mean powered in a loop? 

Also did you remember to expand the file system via raspi-config?

beefsnaps, 2015-04-26:
Hi All,

Right, I've bought a bigger power supply - 5v 12a - and wired the rpi and the LED to the +V and -V terminals. Still getting constant colour on the first 18 LEDs and when I fire up Ambi-TV I get flashing of the same LEDs. The bottom row seem to be working fine but the sides are a bit random.

Any ideas? Could it be an error on the LED reel? Could I switch the top row for some of the spare LEDs? Or is it a software problem? I've expanded the file system and think I've installed everything as it should be.

Cheers.

ghgoldberg, 2015-04-26:
Is the usb converter attached to the power supply grounds?

I am sure that you have done this, but I'll say it anyhow: the LED strip signals are unidirectional.  Are you feeding the signal in the proper direction?

G-

Sent on the new Sprint Network from my Samsung Galaxy S®4.

<div>-------- Original message --------</div><div>From: beefsnaps <notifications@github.com> </div><div>Date:04/26/2015  5:23 PM  (GMT-05:00) </div><div>To: gkaindl/ambi-tv <ambi-tv@noreply.github.com> </div><div>Cc: "Geoffrey H. Goldberg" <GHG@GeoffreyGoldberg.com> </div><div>Subject: Re: [ambi-tv] Almost there but getting crazy disco light show (#86) </div><div>
</div>Hi All,

Right, I've bought a bigger power supply - 5v 12a - and wired the rpi and the LED to the +V and -V terminals. Still getting constant colour on the first 18 LEDs and when I fire up Ambi-TV I get flashing of the same LEDs. The bottom row seem to be working fine but the sides are a bit random.

Any ideas? Could it be an error on the LED reel? Could I switch the top row for some of the spare LEDs? Or is it a software problem? I've expanded the file system and think I've installed everything as it should be.

Cheers.

ghgoldberg, 2015-04-26:
Also, have you checked your config files? Are the LEDs properly identified?

Sent on the new Sprint Network from my Samsung Galaxy S®4.

<div>-------- Original message --------</div><div>From: beefsnaps <notifications@github.com> </div><div>Date:04/26/2015  5:23 PM  (GMT-05:00) </div><div>To: gkaindl/ambi-tv <ambi-tv@noreply.github.com> </div><div>Cc: "Geoffrey H. Goldberg" <GHG@GeoffreyGoldberg.com> </div><div>Subject: Re: [ambi-tv] Almost there but getting crazy disco light show (#86) </div><div>
</div>Hi All,

Right, I've bought a bigger power supply - 5v 12a - and wired the rpi and the LED to the +V and -V terminals. Still getting constant colour on the first 18 LEDs and when I fire up Ambi-TV I get flashing of the same LEDs. The bottom row seem to be working fine but the sides are a bit random.

Any ideas? Could it be an error on the LED reel? Could I switch the top row for some of the spare LEDs? Or is it a software problem? I've expanded the file system and think I've installed everything as it should be.

Cheers.

beefsnaps, 2015-04-27:
Hi G,

What do you mean USB converter? Do you mean the USB hub? It is plugged into the DC adapter in the wall. How can I connect the grounds?

LED cables are connected to the bottom of the one with the arrow going away from it, assume this is correct. The power and ground are connected in in a circuit from the end of the reel back to the start again.

The ambi config file is set up correctly I think from counting the LEDs - the LEDs that light up correctly seem to correspond to what is happening on the screen.

Thanks Matt

ghgoldberg, 2015-04-27:
I meant the video-to-usb converter.  But you should power the usb hub also from your 5V power supply.  Common grounds everywhere.

 

G-