gkaindl/ambi-tv #94: 4K, HDCP 2.2 Support

bobobado, 2018-02-25:
First of all, I love this project. I can’t even imagine watching my tv without it anymore, which brings me to my issue: I recently upgraded my TV and ran into the dreaded HDCP 1.4 V HDCP 2.2 compatibility issue when trying to push 4K as I had an older receiver with a new TV. I upgraded my audio receiver to correct this and since then, am unable to get ambi-tv to function with my 4K sources. My HD sources work correctly but not the new, HDCP 2.2 sources. I attempted to resolve this by getting an HDCP converter shown here: https://www.monoprice.com/product?c_id=101&cp_id=10114&cs_id=1011414&p_id=15242&seq=1&format=2. Unfortunately this didn’t work. If I dial down the resolutions of my sources to 1080, ambi-tv works again but never with 4K. My setup is a Sony STR-DH770 receiver connected to this splitter https://www.amazon.com/SIIG-Premium-HDMI-Splitter-18Gbps/dp/B00YT6V9A6/ref=sr_1_31?ie=UTF8&qid=1519578431&sr=8-31&keywords=4K+HDMI+splitter then directly to the TV and then into the converter mentioned above, then into the HDMI to RCA converter and finally, into a STK1160 chipset USB Grabber. I know this project hasn’t been updated in a while but any help is really appreciated. Thank you

ghgoldberg, 2018-03-17:
I also love my ambilight system. I don't have a 4k system yet, but that is coming. Right now, I have a Panasonic 65" plasma that is professionally calibrated.

Check out my hardware build in this thread below. My ambilight processor hardware looks like a professional unit. If you want to team up on getting this to work at 4k, I'm all in. My software skills aren't great, but with hardware and wiring I am top notch.

bobobado, 2018-03-17:
Hello ghgoldberg! I meant to mention that I actually got my setup working! My issue was that I needed an active scaler for the Ambilight side. I did some research and many people have had success with this https://www.hdfury.com/product/linker-4k60-444-600mhz-scaler/ but I was hoping to find a cheaper solution as you still need a splitter with this device. After doing some more digging, I found this on amazon EZCOO 4K HDMI 2.0 Splitter HDR Scaler, HDMI Splitter 1x2 4K 60Hz (4:4:4) HDCP 2.2, 4K & 1080P Scaler Output Together, SPDIF Audio DE-embed,Dolby Atmos, DTS:X,HDMI 1 in 2 out Splitter https://www.amazon.com/dp/B078YRDZKJ/ref=cm_sw_r_cp_tai_cuvRAb3X39DT6. This worked! I had to do some color adjustments in my config file however. I’m assuming it has to do with HDR and color format but it passes full 4K @ 60hz, HDR and 4:4:4 color to my television and 1080p to my RCA converter. I’m not sure why it’s no longer available on Amazon but maybe it can be found elsewhere. Anyway, I hope this helps and good luck!