gkaindl/ambi-tv #95: usbtv driver failes to compile

spyhunter99, 2019-03-23:
I'm not entirely sure what's going on here. Running on an rpi 3. Any ideas on how i can resolve this?

````
root@raspberrypi:/home/pi/ambi-tv-master/misc# cd usbtv-driver/
root@raspberrypi:/home/pi/ambi-tv-master/misc/usbtv-driver# make
make -C /lib/modules/4.14.98-v7+/build  M=/home/pi/ambi-tv-master/misc/usbtv-driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.98-v7+'
  CC [M]  /home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.o
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c: In function ‘usbtv_image_chunk’:
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:317:23: error: ‘struct vb2_buffer’ has no member named ‘v4l2_buf’
                buf->vb.v4l2_buf.field = V4L2_FIELD_INTERLACED;
                       ^
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:318:23: error: ‘struct vb2_buffer’ has no member named ‘v4l2_buf’
                buf->vb.v4l2_buf.sequence = usbtv->sequence++;
                       ^
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:319:40: error: ‘struct vb2_buffer’ has no member named ‘v4l2_buf’
                do_gettimeofday(&buf->vb.v4l2_buf.timestamp);
                                        ^
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c: At top level:
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:635:23: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
        .queue_setup = usbtv_queue_setup,
                       ^~~~~~~~~~~~~~~~~
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:635:23: note: (near initialization for ‘usbtv_vb2_ops.queue_setup’)
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:638:26: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
        .stop_streaming = usbtv_stop_streaming,
                          ^~~~~~~~~~~~~~~~~~~~
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:638:26: note: (near initialization for ‘usbtv_vb2_ops.stop_streaming’)
In file included from ./include/linux/bitops.h:38:0,
                 from ./include/linux/kernel.h:11,
                 from ./include/linux/list.h:9,
                 from /home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:32:
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c: In function ‘usbtv_probe’:
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:719:16: error: ‘V4L2_FL_USE_FH_PRIO’ undeclared (first use in this function)
        set_bit(V4L2_FL_USE_FH_PRIO, &usbtv->vdev.flags);
                ^
./arch/arm/include/asm/bitops.h:183:42: note: in definition of macro ‘ATOMIC_BITOP’
 #define ATOMIC_BITOP(name,nr,p)  _##name(nr,p)
                                          ^~
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:719:8: note: in expansion of macro ‘set_bit’
        set_bit(V4L2_FL_USE_FH_PRIO, &usbtv->vdev.flags);
        ^~~~~~~
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:719:16: note: each undeclared identifier is reported only once for each function it appears in
        set_bit(V4L2_FL_USE_FH_PRIO, &usbtv->vdev.flags);
                ^
./arch/arm/include/asm/bitops.h:183:42: note: in definition of macro ‘ATOMIC_BITOP’
 #define ATOMIC_BITOP(name,nr,p)  _##name(nr,p)
                                          ^~
/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.c:719:8: note: in expansion of macro ‘set_bit’
        set_bit(V4L2_FL_USE_FH_PRIO, &usbtv->vdev.flags);
        ^~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.o' failed
make[2]: *** [/home/pi/ambi-tv-master/misc/usbtv-driver/usbtv.o] Error 1
Makefile:1535: recipe for target '_module_/home/pi/ambi-tv-master/misc/usbtv-driver' failed
make[1]: *** [_module_/home/pi/ambi-tv-master/misc/usbtv-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.98-v7+'
Makefile:6: recipe for target 'all' failed
make: *** [all] Error 2
root@raspberrypi:/home/pi/ambi-tv-master/misc/usbtv-driver#

````

dedeibel, 2019-03-24:
I think with current raspbian versions there is no need anymore for custom modules. Just try `modprobe usbtv` on your system.

spyhunter99, 2019-03-25:
gotcha, after playing around with it for a few hours (motion), i then tested with vlc which works reliably. Long story short my issue was that motion doesn't like the usbtv adapter.

closed 2019-03-25 by spyhunter99